tierneytim/Pico-USB-audio #9: Audio Input

nikitalita, 2022-04-06:
Does this support Audio input?

tierneytim, 2022-04-06:
No, but depending on what you need that might be possible.

By audio input do you mean support for interfacing with 12S/PDM mics? 
I probably won's support that. 

But if you want to transfer digital audio from the rp2040 to a laptop over USB that could be made possible. 

What project do you have in mind?

nikitalita, 2022-04-06:
I2S from an ADC to USB. So a USB Audio class device.

tierneytim, 2022-04-07:
The USB part should be possible. What sample rates are you hoping to achieve?

nikitalita, 2022-04-07:
24-bit 44.1khz or 48k. 

the real thing is, I've found a micro controller that is absolutely perfect for this, it has all this built in and is also programmable and it has GPIO ports to communicate with other devices (like a MIDI interface): https://www.cmedia.com.tw/products/USB20_FULL_SPEED/CM6530N

But there's ZERO documentation on how to program this and trying to get it from the company is like pulling teeth. So I'm looking for backups.

tierneytim, 2022-04-10:
USB transfer to a computer is now functional. There is a[ test case](https://github.com/tierneytim/Pico-USB-audio/blob/main/SDM/examples/record/record.ino) in the examples with a 480Hz sine wave.  Just open up audacity and select the mbed device as your microphone and it will hopefully just work.

closed 2022-04-15 by tierneytim