tierneytim/Pico-USB-audio #14: Compiler errors

w9ran, 2022-08-19:
I've installed the mbed support and use it often to compile other software for the Pico,  but after I downloaded the zip file from Github and tried using the "Add zip library" method, the IDE said the file did not include a valid library.    I then tried to add it manually by extracting Pico-USB-Audio-main.zip into the Arduino/libraries folder.    But the compilation of tone.ino fails with the message: "
                    pdm_pio_program_get_default_config
exit status 1
'set_sys_clock_khz' was not declared in this scope

Which sounds like it's not able to find some needed files from the sdk, even though I've had no problem compiling other Pico code.  
I tried compiling mbedUSB which fails with "pdmAudio.h: No such file or directory" so the IDE is not finding the required library files.   I know this has been a longstanding issue with the Arduino IDE but their Library Manager has pretty much fixed it and some of the information I found online about manual installation of libraries is outdated and not supported by the current IDE version.

I suspect I'm doing something wrong as I'm pretty new to the Pico but would appreciate any assistance resolving this as I'm anxious to try out the USB audio.   Thanks!

Bob

tierneytim, 2022-08-19:
Hey, the library is the SDM folder not the pico-usb-audio folder. Try putting that folder in your libraries folder

w9ran, 2022-08-19:
Got it right now, thanks.   But a new head-scratcher:  it compiles without error and downloads to the Pico fine, but that's it.  All I hear (using tone.ino example) is a click when power is applied, no tones.   Scoping pin 14 shows it just sitting near 0V except for that noise spike at power on.    Tried a new Pico, same thing.    Any idea what might be going on?

Bob

tierneytim, 2022-08-20:
I updated my Arduino and the mbed to the latest version and the same thing happened to me although the mbedUSB exampled worked the tone example didn't. I've moved the filling in of the lookup table to the begin method and that seemed to work so I've committed that. Let me know if that fixes things on your side as well.

w9ran, 2022-08-20:
Aha!  It works for me now with the updated library.  I'd skipped the tone example and had success using mbedUSB and the record example so figured it was something unique to the tone code.     I'm mostly interested in learning to send audio from the Pico to the PC so the record example is very helpful, thanks.   The PDM audio does sound pretty good.

For some reason I've started having trouble uploading code to the Pico from the IDE with the dreaded "the selected serial port does not exist" error.   This is frustrating because the complier doesn't seem to create the uf2 file that could be used to transfer the code manually.    It's like the serial port gets locked even though it doesn't show up in Device Manager.   Not your problem thought!

tierneytim, 2022-08-20:
I think it is my fault. When you use the USB audio you disable serial. Just hold down boot while resetting. Anyway I'll consider this closer

closed 2022-08-20 by tierneytim