tierneytim/Pico-USB-audio #16: "USB microphones" (example record.ino) does not work on a Pi Pico 2020 (macOs)

mgiugliano, 2022-09-29:
The Pi works (by the blink example) and the "record.ino" code compiles and flashes without errors.

A USB interface does appear both as an Input as well as an Output device, named "Mbed Audio".

Despite the generation of the sinusoid, nothing is "heard" via Audacity or via the vu-meter software monitor of the OS System Preference selection panel.

Could it be a matter of input amplitude? Any help would be precious to me.

mgiugliano, 2022-09-29:
Unless I am really wrong, the following test sketch reveals that `pdm.sine_lu(600)` always returns zeros! 

``` C++
#include "pdmAudio.h"

pdmAudio pdm;

void setup() {
  Serial.begin(115200);
}

void loop() {
  int16_t left = pdm.sine_lu(600);
  Serial.println(left);
}
```

Is it intended? Where am I wrong? I hope it helps.

mgiugliano, 2022-09-30:
This sketch generates (white) noise and it plays inside the PC! This proves that the library works fine.
Could this be maybe added as a quick and dirty additional example?

```c++

#include "pdmAudio.h"

pdmAudio pdm;

void setup() {
 pdm.USB_UAC();
 randomSeed(123);
}

void loop() {
  int16_t  left = random(32767);
  int16_t right = left;

  // write  stereo data to computer
  pdm.USBtransfer(left, right);
}
```

Instead this other sketch does not produce a sinusoidal tone as expected:

```c++

#include "pdmAudio.h"

pdmAudio pdm;

const uint16_t sampleRate = 48000; // Hz
float Dt = 1./sampleRate;         // s
float freq = 500.;                // Hz
float T    = 1./freq;             // s

float t    = 0.;                  // s

void setup() {
 pdm.USB_UAC();
}

void loop() {
  t = (t > T) ? 0 : t + Dt;
  int16_t left = (int16_t)(sin(2.0 * PI * freq * t) * 32767 * 1);
  int16_t right = left;
  // write  stereo data to computer
  pdm.USBtransfer(left, right);
}
```

It must be something about the MSB or LSB binary format of int16_t.....

tierneytim, 2022-09-30:
Thanks for all the detective work. I'll have a look at this on Sunday and report back

tierneytim, 2022-10-17:
ok, This should be working now. I moved the initialization of the lookup table to the begin method which now needs to be called for the example to work. example is now udpated. let me know if this works

closed 2022-10-20 by tierneytim