tierneytim/Pico-USB-audio #17: Raspberry Pico USB support

lu7did, 2022-10-25:
Hello, very good implementation.
I wonder why the USB read functions are left out of the builting if working with a Raspberry Pico board.
Can you comment on that?
Regards, Pedro.

tierneytim, 2022-10-25:
Hi, it doesn't have anything to do with the board. It is the software. The Arduino mbed core should happily run on any RP2040-based board and allow for USB reading and writing. The reason I support the mbed approach is that I prefer developing in the Arduino environment. 

I previously had a version based purely on the pico-sdk, but as the SDK evolved and the official USB stack changed to tinyUSB this became too difficult to maintain. 

With regards to the unofficial Arduino core for RP2040 boards, it does not have support for UAC  drivers and therefore can't be used with this library.  You can play high-quality tones but that would be it. 

It is a long-term goal to get the tinyUSB code working so that this can generalise to other microcontrollers but this has proved challenging so far.

lu7did, 2022-10-25:
Thank you very much for your kind answer.

I'm actually developing using the Arduino IDE with the core created by Earle F. Philhower, III which is reasonably functional for my project. Actually it's more than the core software as a number of libraries has been ported to that core specifically for the support of the Arduino IDE and I'm not certain on the impact it might have to migrate to another core in the middle of the project.

My needs are very simple, I do need to receive a digital data stream from a host computer over USB and be able to process it, the board should look like a sound card for that host. No playing of that data needed at all (it's played but by other means which aren't audio).

On a luxury level I would need to get tone information and sent over to the host as if it was captured by a microphone.

Any thoughts on how your implementation can be used for that or pointers in that overall direction?

Thanks in advance for your time and patience.

Regards, Pedro.

tierneytim, 2022-10-26:
I think without migrating to the official mbed core (which I appreciate now would be challenging) the only option is to try and get this [example](https://github.com/raspberrypi/pico-playground/tree/master/apps/usb_sound_card) working. that will solve the data transfer issue of the computer to rp2040. However, I don't think that has the ability to transfer data back to the computer. For that, you would need to look at this [repository](https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico).  I'm going to close this issue now as it overlaps with the existing issue referenced in my first answer. Sorry, I can't be more helpful. And best of luck with your project.

closed 2022-10-26 by tierneytim